thinkoco/c5soc_opencl #3: OpenCL SDK License issue in Quartus 16.1 lite edition (DE1_SoC)

ArjunaDeSoysa, 2018-01-22:
I installed the Quartus 16.1 lite edition(No License required). But in compiling process(make .aocx file) it asking for license file. Some one please help me to solve this problem.  
[image]

thinkoco, 2018-01-29:
Yes，Quartus lite edition you need an opencl license to support it working.

thinkoco, 2018-01-29:
@ArjunaDeSoysa Quartus 16.1 lite edition no need Quartus license and contains some  free IPs.When compiling opencl project ，you need an opencl license.

thinkoco, 2018-01-30:
@ArjunaDeSoysa  Maybe,you can ask your teacher to apply for a trial license form terasic. I'm not sure that you can get  one,but you can try it.

ArjunaDeSoysa, 2018-01-30:
Ok I'll try it..thank you very much for replying....

closed 2018-02-01 by thinkoco