thinkoco/c5soc_opencl #4: Support for Intel OpenCL SDK 17.1

hammadj, 2018-03-02:
Hi there, I am trying to get this to work with OpenCL SDK 17.1 on my de10-nano board, but am having trouble. I am trying to run this project: https://github.com/doonny/PipeCNN.

I was able to successfully get it working using the OpenCL image for the de10 nano provided by terasic, but unfortunately that image doesnt support HDMI and ethernet. I cross-compiled the PipeCNN kernel and host using Intel OpenCL SDK 17.1 and the [de10-nano BSP](http://www.terasic.com.tw/cgi-bin/page/archive.pl?Language=English&No=1046&PartNo=4) on my desktop and transferred to the de10-nano board. It successfully ran there, even though it had the 16.1 binaries.

I then tried using this repo's BSP and followed the [instructions here](https://github.com/thinkoco/c5soc_opencl/blob/master/HowToDo.md), recompiling the PipeCNN kernel with the de10_nano_sharedonly_hdmi BSP from this repo. However, it would give this error: 
`libintel_soc32_mmd.so cannot opened shared object file: no such file or directory`
and various other errors. Im guessing this error is because that file used to be called `libalterammdpcie.so` but intel changed a bunch of the names in 17.1. So, I tried recompiling the kernel drivers for 17.1 (zImage, socfpga.dtb, and opencl.rbf) as well as getting the latest aocl binaries from intel for 17.1. I was finally able to get PipeCNN to run, however the execution gets stuck on layer 1. You can read more about that here: https://github.com/doonny/PipeCNN/issues/53.

Do you know what steps I'm missing to get this working with 17.1? Unfortunately I cant do 16.1 because of licensing issues that were not able to be resolved with the eth0 fix. Intel removed the requirement for licensing in 17.1 when using aoc, so thats the only choice I have.

thinkoco, 2018-03-02:
@hammadj  I think it's my fault. I'm checking it now. Maybe, there are something wrong with my Linux image. I will update a image for you later. 
As my license is expiration, I can't compile the kernel for testing. Can you do this work for me?
Recompiling the PipeCNN kernel  with my de10_nano_sharedonly_hdmi BSP ,the run on the terasic de10_nano opencl linux image. ( repalce opencl.rbf and cnn.aocx)  See if it works without desktop by my BSP.

hammadj, 2018-03-02:
@thinkoco 
Ok I will test that config out and let you know in a few hours. Thanks for your help, its much appreciated!

thinkoco, 2018-03-02:
@hammadj Thank you for mentioning the license issue. 
You can try this [ linux image](https://drive.google.com/file/d/1ReJ8vTKqo6uupvrX6Otb3pLFxlG-8fxc/view?usp=sharing) . source the **int_opencl.sh** and run **boardtest** in /root folder. If it's ok, then you can update the **aocl-rte-16.1.0-1.arm32** folder  wtih the **aocl-rte-17.1.0-590.arm32.tgz** and change the path in  int_opencl.sh  file. Also upadte the driver and zImage,replace the rbf and run PipeCNN. If it get something wrong, please let me know. Thanks.

hammadj, 2018-03-02:
@thinkoco 

Ok so I flashed that linux image, sourced the ini_opencl.sh, and ran the boardtest. It got stuck at this part: 
[image]

Also, I tried what you said in your previous comment, compiling PipeCNN with your BSP and running it on the no-desktop version from terasic, and I got the same error: `libintel_soc32_mmd.so cannot opened shared object file: no such file or directory`

thinkoco, 2018-03-03:
@hammadj Maybe ,I had build the boardtest with wrong command,so it get this result.
No need to care about DE10-STANDARD string.
Here are some 17.1 [binaris ](https://drive.google.com/open?id=1JlzIWzoqYgqMxz6cxRFaPa6ahCgs849Z)that i fix the libintel_soc32_mmd.so error.
you can update the sd card image(180302.img)  wtih these files, then sudo apt-get install libsdl2-dev and  run the host.run.
I'm very sorry that there is no board to test these binaris now.

thinkoco, 2018-03-09:
@hammadj The BSP is OK , but there is something wrong with my sd card iamge. I have fixed this issue.U can run with the latest SD card image [here](https://drive.google.com/file/d/1uZkOzOZokQ7RGGJ8iyFA7wrLdqCmZTSu/view?usp=sharing) . Read the readme.txt in sd card first！
opencl runtime is 17.1
[image]

nocduro, 2018-03-09:
@thinkoco Thank you for your help on this!

I've been working with Hammad on it, and your updated SD card image is working on our board now!

nocduro, 2018-03-15:
@thinkoco How did you fix your linux image? I'm trying to get a webcam to work with your new image, but it doesn't look like the webcam drivers were enabled when the image was built? I tried using my own zImage that I compiled with webcam support, but then when I run PipeCNN it freezes like before.

thinkoco, 2018-03-16:
@nocduro  Hi，
The zImage in my sd card image contains uvc driver and colorApp is OK.  Also, you can checkout  the 3.18 branch in my linux-socfpga repo to rebuild zImage.  
```
git checkout -b socfpga-opencl_3.18 origin/socfpga-3.18
cp config_opencl_de10_nano .config

export ARCH=arm
export CROSS_COMPILE=arm-linux-gnueabihf-
export LOADADDR=0x8000

make zImage
```
After this, remember to rebuild the OpenCL driver.
copy the runtime foldr to PC, go to the driver folder
```
make KDIR=../(to linux-socfpga folder)
```
Then, copy the aclsoc_drv.ko back to sd card and cover the original one.

roddregs, 2018-04-16:
Hi @thinkoco .
I would like to experiment running PipeCNN using De0-Nano-SoC board.
Do you think the configurations of PipeCNN will smoothly run in De0-Nano-SoC board?
Thanks in advance.

thinkoco, 2018-04-18:
@roddregs  Hi, You can compile the PipeCNN kernel with de0-nano-soc opencl bsp and check the report. I'm not sure if the minimum configuration of PipeCNN will  be compiled successfully.There are just 40k LEs in de0-nano-soc(85k LEs in de1-soc, 110k LEs in de10-nano or standard).
[image]

roddregs, 2018-04-19:
@thinkoco 
Hi Sir, good day!
Tried to compile PipeCNN in Deo-Nano-SoC board.
However, I encountered the following problem when running the main makefile:
g++ ./host/main.o ../common/ocl_util.o ../common/timer.o -o run.exe -L/thesis/intelFPGA/16.1/hld/board/de1soc/arm32/lib -L/thesis/intelFPGA/16.1/hld/host/arm32/lib -L/thesis/intelFPGA/16.1/hld/host/linux64/lib -Wl,--no-as-needed -lalteracl -lalterahalmmd -lalterammdpcie -lelf
/usr/bin/ld: cannot find -lalterammdpcie
collect2: error: ld returned 1 exit status
Makefile:118: recipe for target 'run.exe' failed
make: *** [run.exe] Error 1
Is this an indication that De0-Nano is not compatible with the PipeCNN configurations?
Thank you very much for your reply in advance, Sir @thinkoco ...

thinkoco, 2018-04-20:
@roddregs  It's no finding  alterammdpcie library ,so you can check the OpenCL environment that you have set. . The limit may be fpga logic resources, not the host.So,you can build the OpenCL kernel first and check the resource utilization in "acl_quartus_report.txt" or some other report files.

roddregs, 2018-06-02:
@thinkoco ..
I am now using DE1-SoC board...
When running main makefile provided, I got the following error:
[image]
Please, can you suggest any solutions to this error I encountered?...
Thank you very much...

thinkoco, 2018-06-02:
Hi,usearm-linux-gnueabihf-g++ 4.9 or 4.8Best Regards!Knat-------- 原始邮件 --------主题：Re: [thinkoco/c5soc_opencl] Support for Intel OpenCL SDK 17.1 (#4)发件人：roddregs 收件人：thinkoco/c5soc_opencl 抄送：Welon Tank ,Mention @thinkoco ..
I am now using DE1-SoC board...
When running main makefile provided, I got the following error:

Please, can you suggest any solutions to this error I encountered?...
Thank you very much...

{
"@type": "MessageCard",
"@context": "http://schema.org/extensions",
"hideOriginalBody": "false",
"originator": "37567f93-e2a7-4e2a-ad37-a9160fc62647",
"title": "Re: [thinkoco/c5soc_opencl] Support for Intel OpenCL SDK 17.1 (#4)",
"sections": [
{
"text": "",
"activityTitle": "**roddregs**",
"activityImage": "https://assets-cdn.github.com/images/email/message_cards/avatar.png",
"activitySubtitle": "@roddregs",
"facts": [

]
}
],
{
"name": "Add a comment",
"@type": "ActionCard",
"inputs": [
{
"isMultiLine": true,
"@type": "TextInput",
"id": "IssueComment",
"isRequired": false
}
],
"actions": [
{
"name": "Comment",
"@type": "HttpPOST",
"target": "https://api.github.com",
"body": "{\n\"commandName\": \"IssueComment\",\n\"repositoryFullName\": \"thinkoco/c5soc_opencl\",\n\"issueId\": 4,\n\"IssueComment\": \"{{IssueComment.value}}\"\n}"
}
]
},
{
"name": "Close issue",
"@type": "HttpPOST",
"target": "https://api.github.com",
"body": "{\n\"commandName\": \"IssueClose\",\n\"repositoryFullName\": \"thinkoco/c5soc_opencl\",\n\"issueId\": 4\n}"
},
{
"targets": [
{
"os": "default",
"uri": "https://github.com/thinkoco/c5soc_opencl/issues/4#issuecomment-394057803"
}
],
"@type": "OpenUri",
"name": "View on GitHub"
},
{
"name": "Unsubscribe",
"@type": "HttpPOST",
"target": "https://api.github.com",
"body": "{\n\"commandName\": \"MuteNotification\",\n\"threadId\": 308610385\n}"
}
],
"themeColor": "26292E"
}

roddregs, 2018-06-02:
@thinkoco 
Thank you very much @thinkoco, it solved the problem... 

However, I got another problem when running main makefile : 
After compiling, it says:
Error: cannot fit kernel(s) on device

[image]

The screenshot of the flow summary is the following: 
[image]

Please, any suggestion on how to solve this error?
Thank you very much!

thinkoco, 2018-06-02:
Hi,you can disable the  "--profile" flagsin Makefile and rebuild it.Best Regards!Knat

roddregs, 2018-06-02:
Thanks @thinkoco, I did it and it worked... Thanks soo much...

However, after the compilation, I have another error when I run the command: ./run.exe conv.aocx

[image]

Kindly help me how to solve this, please @thinkoco ... Thanks very much...

thinkoco, 2018-06-02:
Hi，so，you may need to run it on de1soc，not on PCBest Regards!Knat

roddregs, 2018-06-05:
Hi @thinkoco ...
Thank you very much for your kind assistance, it worked....
                             
root@socfpga:~# source ./init_opencl.sh                                         
root@socfpga:~# aocl diagnose                                                   
aocl diagnose: Running diagnostic from /home/root/opencl_arm32_rte/board/c5soc/n
                                                                                
Verified that the kernel mode driver is installed on the host machine.          
                                                                                
Using platform: Altera SDK for OpenCL                                           
Board vendor name: Altera Corporation                                           
Board name: de1soc_sharedonlyCyclone V SoC Development Kit                      
                                                                                
Buffer read/write test passed.                                                  
                                                                                
DIAGNOSTIC_PASSED                                                               
root@socfpga:~# ls                                                              
PipeCNN  README  init_opencl.sh  opencl_arm32_rte  swapper                      
root@socfpga:~/PipeCNN# aocl program /dev/acl0 conv.aocx
aocl program: Running reprogram from /home/root/opencl_arm32_rte/board/c5soc/arn
Reprogramming was successful!                                                   
root@socfpga:~/PipeCNN# aocl diagnose                                           
aocl diagnose: Running diagnostic from /home/root/opencl_arm32_rte/board/c5soc/n
                                                                                
Verified that the kernel mode driver is installed on the host machine.          
                                                                                
Using platform: Altera SDK for OpenCL                                           
Board vendor name: Altera Corporation                                           
Board name: de1soc_sharedonlyCyclone V SoC Development Kit                      
                                                                                
Buffer read/write test passed.                                                  
                                                                                
root@socfpga:~/PipeCNN# ./run.exe conv.aocx
***************************************************                             
PipeCNN: An OpenCL-Based FPGA Accelerator for CNNs                              
***************************************************                             
                                                                                
Platform: Altera SDK for OpenCL                                                 
Using 1 device(s)                                                               
  Device 0: de1soc_sharedonlyCyclone V SoC Development Kit                      
Device OpenCL Version: OpenCL 1.0 Altera SDK for OpenCL, Version 16.0           
Device Max Compute Units: 1                                                     
Device Max WorkGroup Size: 2147483647                                           
Device Max WorkItem Size: 2147483647                                            
Device Global Memory Size: 512 MBytes                                           
Device Local Memory Size: 16 KBytes                                             
Device Max Clock Freq: 1000 Mhz                                                 
                                                                                
Loading kernel/binary from file conv.aocx                                       
Reprogramming device with handle 1                                              
                                                                                
61063552 total weights read                                                     
1024 total output reference read                                                
                                                                                
                                                                                
154587 bytes image data read from binary files                                  
                                                                                
Executing Layer 1:                                                              
                                                                                
Launching single work-item kernel winbuffer                                     
                                                                                
Launching single work-item kernel Conv                                          
                                                                                
Launching single work-item kernel Pooling                                       
                                                                                
Launching kernel MemWr with local size: 1, 1, 8  (global size: 27, 27, 96)      
                                                                                
Launching kernel lrn with local size: 1, 1, 12  (global size: 27, 27, 12)       
                                                                                
Executing Layer 2:                                                              
[... 107 lines cut ...]
[... cut ...]
    MemWr: 1.574 ms                                                             
    Lrn  : 0.000 ms                                                             
                                                                                
Total kernel runtime 154.403 ms                                                 
Batch size = 1, average process time per batch: 154.403 ms                      
                                                                                
Total runtime: 0.159448s

roddregs, 2018-06-05:
Hi @thinkoco ..
I am planning to experiment the PipeCNN using my own dataset. I am thinking of using Caffe for training my network model..
Do you have any idea or suggestions on how will I make it successfully?
Thank you very much for always sharing your expertise and time...

Enshall, 2018-06-08:
Dear @thinkoco

thinkoco, 2018-06-08:
Hi,you may add the CL_CONTEXT_COMPILER_MODE_ALTERA=3(sdk16.1)or CL_CONTEXT_COMPILER_MODE_INTELFPGA=3(sdk7.1) in the initial shell file.  Also,you can export the flags before you run host.Best Regards!Knat-------- 原始邮件 --------主题：Re: [thinkoco/c5soc_opencl] Support for Intel OpenCL SDK 17.1 (#4)发件人：Enshall 收件人：thinkoco/c5soc_opencl 抄送：Welon Tank ,Mention Dear @thinkoco
When I  run PipeCNN on de1soc with the image --c5soc_opencl_lxde_all_in_one_180317.img, VGA output will shut down, both direct connect or serial port.
when I compile the bitstream file and host file, I replaced the bsp with your de1soc_sharedonly_vga.
Can you give me some guides?
Best wishes for you!

{
"@type": "MessageCard",
"@context": "http://schema.org/extensions",
"hideOriginalBody": "false",
"originator": "AF6C5A86-E920-430C-9C59-A73278B5EFEB",
"title": "Re: [thinkoco/c5soc_opencl] Support for Intel OpenCL SDK 17.1 (#4)",
"sections": [
{
"text": "",
"activityTitle": "**Enshall**",
"activityImage": "https://assets-cdn.github.com/images/email/message_cards/avatar.png",
"activitySubtitle": "@Enshall",
"facts": [

]
}
],
{
"name": "Add a comment",
"@type": "ActionCard",
"inputs": [
{
"isMultiLine": true,
"@type": "TextInput",
"id": "IssueComment",
"isRequired": false
}
],
"actions": [
{
"name": "Comment",
"@type": "HttpPOST",
"target": "https://api.github.com",
"body": "{\n\"commandName\": \"IssueComment\",\n\"repositoryFullName\": \"thinkoco/c5soc_opencl\",\n\"issueId\": 4,\n\"IssueComment\": \"{{IssueComment.value}}\"\n}"
}
]
},
{
"name": "Close issue",
"@type": "HttpPOST",
"target": "https://api.github.com",
"body": "{\n\"commandName\": \"IssueClose\",\n\"repositoryFullName\": \"thinkoco/c5soc_opencl\",\n\"issueId\": 4\n}"
},
{
"targets": [
{
"os": "default",
"uri": "https://github.com/thinkoco/c5soc_opencl/issues/4#issuecomment-395659898"
}
],
"@type": "OpenUri",
"name": "View on GitHub"
},
{
"name": "Unsubscribe",
"@type": "HttpPOST",
"target": "https://api.github.com",
"body": "{\n\"commandName\": \"MuteNotification\",\n\"threadId\": 308610385\n}"
}
],
"themeColor": "26292E"
}

Enshall, 2018-06-08:
Thank you for your quick reply @thinkoco sincerely
It will display error : CL_INVALID_KEENEL_NAME  when I add "CL_CONTEXT_COMPILER_MODE_ALTERA=3"
the source code shows below:
const char *knl_name_memRd = "memRead";
knl_memRd[i] = clCreateKernel(program, knl_name_memRd, &status);
my compiler is 16.1 and i init it with 16 when i run the host
can you give some more advices !

thinkoco, 2018-06-08:
@Enshall  Hi, You may need to  update the rbf file by the top.rbf  which you build.Also the aocx and host need to be updated. You can do as [follow steps](https://github.com/thinkoco/c5soc_opencl#run-opencl-application)

roddregs, 2018-06-08:
Hi @thinkoco 

As reflected in the post above. I was using DE1-SoC Cyclone V board and successfully compiled and run in the board the PipeCNN using PLATFORM=arm32 and flow=hw which only read ONE IMAGE file. Thanks for your assistance on that matter...

And now, I tried to enable OpenCV to experiment loading multiple images from Imagenet 2012 validation set. I set USE_OPENCV = 1 in the Makefile and installed OpenCV 2.4.9 in my PC (Ubuntu 16.04). When I compiled, I got the following error that says:

**libopencv_core.so :file not recognized: File format not recognized**

[image]

Any suggestion on how to solve this error? Is OpenCV 2.4.9 not compatible with the current set-up of PipeCNN?

Thank you very much for always answering my queries...

thinkoco, 2018-06-09:
@roddregs You can install the opencv in the sd card image and build the host on de1-soc.
Aslo,you amy need to modify the OCV_INCLUDES and OCV_LIBDIRS in the Makefile.

Enshall, 2018-06-09:
Thank you for your patiants ! @thinkoco
Actually I can`t  understand step 2 all the time.. Whats the "Windows fat32 partition" means ?
And  how to get a rbf file. I only downloaded the dtb file from your Google Drive.
Best wished！

thinkoco, 2018-06-09:
@Enshall When you building the kernel (make fpga), there will be a top.rbf file generated in the `conv` folder. The sdx1 is the fat32 partition which contains the rbf, zImage and dtb.
[image]

Enshall, 2018-06-11:
Hi @thinkoco , thank you a lot ! I fixed the error of 'CL_INVALID_KERNEL_NAME ' as what you taught me.
You do a really great work here ! Thank you !

roddregs, 2018-06-11:
Hi @thinkoco 
I used **minicom** in Ubuntu 16.04 to run and access de1-soc board.  
You suggested to install OpenCV in sd card image and build the host on de1-soc. Please help me how to do it in since I used the linux_sd_card_image.img from the de1-soc BSP downloaded from Terasic... I don't know how to access other partitions of my sd card (I have 32GB) since i'ts already configured automatically by the  linux_sd_card_image.img. Is minicom not appropriate for this purpose? will i have to use other environment like LXDE?

thinkoco, 2018-06-13:
@roddregs Hi,
1. For sending files to de1soc,you can mount you sd card on Ubuntu PC and copy them to sd card ext4 partition, such as /media/rod/xxxxxxx/root
2. For building host with opencv libs, you can cross compile the opencv on PC and build the host with the cross-compiled opencv libs. Then copy host and opencv libs to sd card. Also, you can copy the source code of opencv libs and pipeCNN to sd card,then build them on de1soc. Actually, I had not run PipeCNN with OpenCV before. So,there are no detailed steps here. You may try these work yourself or ask doonny for help.
3.  Some opencv applications  can run without gui. So, you may check that if the PipeCNN works without gui.

thinkoco, 2018-06-26:
@roddregs You can build the PipeCNN host with opencv libs on DE1SOC .
1. copy the PipeCNN source code to all_in_one_180317 sd card.
2. Power on DE1SOC with the all_in_one_180317 image sd card.
3. source init_opencl_17.1.sh
4. modify the Makefile like this:
[image]
5. make host

Also, you may change the `picture_file_path_head` path in main.cpp.
and  `export DISPLAY=:0` before running host in minicom terminal.

closed 2018-06-26 by thinkoco

pratik18v, 2018-07-13:
Hi @thinkoco ,
I am trying to run the PipeCNN code on the de10 standard board using the image (c5soc_opencl_lxde_all_in_one_180317.img) and BSP you have provided. I was able to successfully compile the code but I am facing an issue similar to an earlier comment in this thread.
When I run the command aocl program /dev/acl0 conv.aocx , the display turns off and I loose all connections (serial/SSH) to the board. Your previous fix (CL_CONTEXT_COMPILER_MODE_ALTERA=3) is already included in the init file, but its not helping. 

Do you know what may cause it and/or how to solve it?
Thank you

thinkoco, 2018-07-14:
After building the PipeCNN，you will get a top.rbf.Then,update the opencl.rbf with the top.rbf and keep the name 'opencl.rbf'. the aocl program is no need. just source the initial file and run host directly.

rahimijan, 2019-09-06:
@hammadj 
please help me for this issue.

rahimijan, 2019-09-06:
@thinkoco 
 I downloaded ("#4 (comment) "),

but, its Linux doesn't boot up! In fact, I have just these pictures:

[image]
[image]

Do you know what the problem is? and, can I ask you, 
if it is possible, test Linux of above link for me, please, I need to work with it.
Also, I want to install opencl on 'Linux LXDE', but I don't know How should I do it.
Please, if you know, help me to solve these issues.

Unfortunately, this Linux can't boot up from sd card (in DE10-Nano board), if you know the solution or another way to booting it up, Please let me know, thank you.

sergio14890, 2020-04-16:
Hey guys! 
I have de1soc 17.1 bsp.
Im try compile the pipeCNN , but i have this error:

`c:/Users/sergi/Downloads/PipeCNN-master/project/device/conv_pipe.cl:1096:16: error: attribute takes one argument
__attribute__((max_work_group_size(1,1,LRN_MAX_LOCAL_SIZE))) // (x,y,z)
               ^
2 warnings and 1 error generated.
`

What im can do?

sergio14890, 2020-04-29:
> @thinkoco
> Thank you very much @thinkoco, it solved the problem...
>
> However, I got another problem when running main makefile :
> After compiling, it says:
> Error: cannot fit kernel(s) on device [...]

I have the same error, i dont can solve it.
What you sugered?

sergio14890, 2020-08-05:
> Do you know what may cause it and/or how to solve it?

> After building the PipeCNN，you will get a top.rbf.Then,update the opencl.rbf with the top.rbf and keep the name 'opencl.rbf'. the aocl program is no need. just source the initial file and run host directly.

Hey @thinkoco i have the same error.
But i don't find anyone file "opencl.rbf"

thinkoco, 2020-08-06:
@sergio14890 
1. Hi,you can disable the  "--profile" flags in Makefile and rebuild the opencl kernel
2. then, you will get a top.rbf, rename it to opencl.rbf and copy it to micro sdcard partition 1 (FAT32 file system partition)

sergio14890, 2020-08-06:
[image]

Hey @thinkoco , when i do it, my fpga crashed :(

sergio14890, 2020-08-06:
> @sergio14890
> 
> 1. Hi,you can disable the  "--profile" flags in Makefile and rebuild the opencl kernel
> 2. then, you will get a top.rbf, rename it to opencl.rbf and copy it to micro sdcard partition 1 (FAT32 file system partition)

Hey @thinkoco thank you so much for your answer!
You really did a great job.

I did what you told me but it didn't work: |

First, i write c5soc_opencl_lxde_fpga_reconfigurable on SD card.
Secound, i copy (top.rbf and rename to opencl.rbf) to fat32 and copy yet de1soc_socfpga.dtb and rename to socfpga.dtb.

But when I turn on the card, the screen remains off and only the putty works.

Do you have any more suggestions?
Thanks for everything, good job

thinkoco, 2020-08-07:
@sergio14890 
Hi, have you set the MSEL[4:0] on your board to 01010 , SW10(1 to 6) on,off,on,off,on,N/A ?
it shows the altera_load error code -4, you may check the opencl.rbf in the sdcard

sergio14890, 2020-08-07:
Yes @thinkoco msel is well configured.
I think it has to do with the .rbf file itself, because I can connect the putty terminal and work well with it, only my screen is still missing the image.

There is no place in the code where I can add the definition service lightdm start before makefile?

sergio14890, 2020-08-07:
@thinkoco could you tell me which bsp you are using for de1soc to generate the .aocx file?
I may be using a different one.
Thanks for everything again

thinkoco, 2020-08-10:
@sergio14890 
 1. the BSP is de1soc_sharedonly_vga, build the opencl kernel with this bsp and replace the opencl.rbf with top.rbf ( which the vip core inside)

 2. [here](https://github.com/thinkoco/c5soc_opencl/blob/master/documents/HowToReconfigureFPGA.md#fpga-reconfiguration) is the cmd to run  desktop

sergio14890, 2020-08-10:
@thinkoco the version of the reconfigurable image does not include opencv?
I tried to transfer the libraries from the latest version but I get errors.

thinkoco, 2020-08-13:
@sergio14890 
sorry, opencv has not been tested on the image.  and i serach the [same error online](https://www.raspberrypi.org/forums/viewtopic.php?t=196070). you can try it, but i'm not sure that it works .

sergio14890, 2020-08-14:
@thinkoco 
I already tried this solution.
Except that there is a problem, the card cannot connect to the internet (prints below).
Is there a way to download the .deb file and transfer it to the card and install there?
[image]
[image]

ngohieutruong, 2020-09-03:
> Hey guys!
> I have de1soc 17.1 bsp.
> Im try compile the pipeCNN , but i have this error:
>
> `c:/Users/sergi/Downloads/PipeCNN-master/project/device/conv_pipe.cl:1096:16: error: attribute takes one [...]

Hi @sergio14890 
I have the same error. Can you guide me to fix it? thanks

datnguyen263, 2021-03-25:
Dear @thinkoco,
i tried follow to issues to fix my problem but i cant solve it. Can u help me with my problem? tks so much!
[image]

timothyljstewart, 2021-03-25:
Does this email reply work? Hmm test test...

Have you checked that you have the correct environment variables exported?

Very Respectfully,

Timothy L. J. Stewart

[image]<https://user-images.githubusercontent.com/55569925/112480182-57ae6080-8da8-11eb-800f-3bc7babf82b2.png>

datnguyen263, 2021-03-26:
> Does this email reply work? Hmm test test... Have you checked that you have the correct environment variables exported? Very Respectfully, Timothy L. J. Stewart
> […](#)

its my email! can u ask me detail environment variables exported?

bulentcandan, 2023-02-16:
> 

Hello,
I am trying to compile example "boardtest" host side on Visiul Studio 2019 and alterammdpcie.lib is not found. I could not find it any where within the OpenCL SDK.  I am using SDK 22.4. Can you provide information on where to get this library. 
Best Regards
Bülent CANDAN