thinkoco/c5soc_opencl #5: Camera on PC

manhha1402, 2018-03-27:
Hi,
I am having a project using Intel FPGA for OpenCL code for DE 10 Standard. I am new to FPGA so I would like to ask some questions
It is possible to connect Camera on usb port on PC, transfer data from PC to DE 10 board for processing, then get back the result and display on PC ?. So I don't need to use HPS on FPGA since I should connect the board to robot, get data from camera on robot and the implementation should be integrated on ROS (Robot Operating System)
Thank you

thinkoco, 2018-03-30:
@manhha1402 hi,
Due to the socfpga based OpenCL environment restriction，when using OpenCL on de10-standard,  the host application must run on HPS. Of course,you can use opencl based on PCI-E interface board,such as DE5-net. the board can be connected to the PC through PCI-E ,  so that you can run  OpenCL host program on PC.
Also,it seems that ROS can be installed on ubuntu. I'm not sure whether the ROS can be installed  into the SD card image of DE10-Standards uccessfully.

manhha1402, 2018-04-12:
Hi,
Thank you for your information. I will try to get familiar to the DE 10 board first, then think about buying DE5-net board ...

closed 2018-04-12 by manhha1402