thinkoco/c5soc_opencl #8: Compile Error with Quartus 17.1（the bsp cannot compatible with）

CrazyBingo, 2018-06-12:
Hello， Thankoco：
     Ths for your achievements about multi fpga kit‘s bsp which support vga and desktop, and I‘m working on DE10-Standard with donny's PipeCNN. 
    After I modified PipeCNN‘s conv_pipe.cl， it can support Quartus 17.1 OpenCL modify and succesfully. 
[image]

Everything seems OK while it goes error wirh "Error: Qsys-script FAILD"， after I refer the conv_pipe.log for details， I have found that it goes wrong while generate Qsys system. So I open the Project of de10_standard_sharedonly_vga,  found that Qsys(Platform Designer) has serveal error, about acl iface interface, the error is like this:

  As you said de10_standard_sharedonly_vga is both supprt 16.1 & 17.1 , but there is sth. wrong while open or compile with 17.1, what's wrong with me, or there is some options that I should set ?
[image]

TKS.  best regards.

thinkoco, 2018-06-13:
@CrazyBingo 您好!
1 如果打开system.qsys,需要先打开 top.qsf project.
2 这个错误可能是因为环境变量造成的. 17.1的环境变量更新为INTELFPGA如:INTELFPGAOCLSDKROOT.为了保证兼容,我将hardware template的script保持和16.1的Platform Designer一致.这样在17.1中使用时,需要设置ALTERAOCLSDKROOT变量.
3 如果您需要在17.1中保持INTELFPGA变量,不使用ALTERA变量,则可以将如/opt/intelFPGA/17.1/hld/board/c5soc/hardware/c5soc_sharedonly/scripts 下的脚本复制替换该项目中的脚本即可.
4 17.1的Platform Designer在编译一些OpenCL 例子确实会出现错误,例如某些版本的boardtest project(这是由于Platform Designer在编译cl时生成的某些同名信号,并将其连接).目前,我还不知道造成这一错误的具体原因.

CrazyBingo, 2018-06-13:
[image]

CrazyBingo, 2018-06-13:
[image]
探讨个更神奇的问题，如上图（Pipe默认参数为例），左边是16.1，右边是17.1 make fpga的，结果左右资源爆表：125%，右边妥妥的88%，神奇，难道16到17 ， altera编译器优化了。。。

CrazyBingo, 2018-06-13:
@thinkoco 
1）前面测试增加ALTERAOCLSDKROOT， 17.1 可以make fpga。
2）顺便测试了不增加环境变量而是修改script，becompare了下主要是以下2个文件中更新为INTELFPGAOCLSDKROOT即可 ，和直接copy c5soc_sharedonly一样。
[image]
但make fpga后测试发现，依然是 Qsys-script FAILD....

thinkoco, 2018-06-13:
@CrazyBingo Hi,
这是供编译前参考估计,编译器本身是不关心这个数据的.即使全部超标,编译也不会在这个阶段终止.具体资源还要看编译后的报告.
这个资源使用率是内核资源和hardware bsp资源对chip资源的占比,  可以在board_spec.xml中得到相关信息,如:
```
  <device device_model="5csxfc6d6f31c8es_dm.xml">
    <used_resources>
      <alms num="1080"/><!-- ALMs used for LUT logic + ALMs used for LUT logic and registers-->
      <ffs num="1908"/>
      <dsps num="0"/>
      <rams num="20"/>
    </used_resources>
  </device>
```
device_model 中有chip资源的描述,而used_resources 则需要编译不带内核的hardware bsp 查看编译后的报告得出,并讲正确的值在这里更新.也就是说,如果这里的used_resources值是错误的,给出的估计利用率偏差会很大.

thinkoco, 2018-06-13:
@CrazyBingo 关于17.1 opencl 需要的环境变量,在altera的手册中有提到过,具体哪个记不太清楚了.
好像17.1中 QUARTUS_ROOTDIR 变为了QUARTUS_ROOTDIR_OVERRIDE
```
export INTELFPGAOCLSDKROOT="/opt/intelFPGA/17.1/hld"
export QSYS_ROOTDIR="/opt/intelFPGA/17.1/quartus/sopc_builder/bin"
export QUARTUS_ROOTDIR_OVERRIDE="/opt/intelFPGA/17.1/quartus"
export PATH="$PATH:${QUARTUS_ROOTDIR_OVERRIDE}/bin:${QUARTUS_ROOTDIR_OVERRIDE}/linux64:${INTELFPGAOCLSDKROOT}/linux64/bin:${INTELFPGAOCLSDKROOT}/bin:${QSYS_ROOTDIR}"
export LD_LIBRARY_PATH="${INTELFPGAOCLSDKROOT}/host/linux64/lib:${AOCL_BOARD_PAKAGE_ROOT}/linux64/lib:${LD_LIBRARY_PATH}"
export AOCL_BOARD_PACKAGE_ROOT="${INTELFPGAOCLSDKROOT}/board/c5soc"
export QUARTUS_64BIT=1
export LM_LICENSE_FILE="/opt/intelFPGA/License.dat"
```

CrazyBingo, 2018-06-13:
@thinkoco : TKS , give you 32 praise. 您这玩SOC FPGA的确实比我这玩裸奔FPGA的来的溜，我再折腾折腾，有问题或者发现，进一步和您探讨，谢谢。

CrazyBingo, 2018-06-13:
早上不知道哪根筋抽风了，用17的scripts跑竟然还是Qsys-script FAILD，下午加了QUARTUS_ROOTDIR_OVERRIDE跑make fpga OK；删除QUARTUS_ROOTDIR_OVERRIDE跑make fpga也OK。
不过继续make host后出现了库的问题。
1）我已经改好了make file中-- to -
2）我已经改好了conv_pipe.c;l中altera为intel等使得支持17.1
3）makefile中增加了 “--arm”（16.1中增加这个可以跑）如下：
COMP_CONFIG = $(shell aocl compile-config --arm) -I$(HOST_INCS)  -DFPGA_DEVICE
LINK_CONFIG = $(shell aocl link-config --arm)
然后还是出错如下，这都最后一步了：
[image]

CrazyBingo, 2018-06-13:
[image]

tzechienchu, 2018-06-13:
if you use in windows please change as follow:
AOCL_LINK_CONFIG := $(wildcard $(ALTERAOCLSDKROOT)/host/arm32/lib/*.so) $(wildcard $(AOCL_BOARD_PACKAGE_ROOT)/arm32/lib/*.so)
But ......
It may compile. But the result won't run on my Terasic OpenCL BSP 
It will give you following error on the SOC Host.
root@socfpga:~/hello_world# ./hello_world
./hello_world: error while loading shared libraries: libalteracl.so: cannot open shared object file: No such file or directory

thinkoco, 2018-06-13:
@CrazyBingo 
aocl link-config --arm show：
```
-LD:\App\intelFPGA\16.1\hld\board\terasic\de1soc\arm32\lib -LD:/App/intelFPGA/16.1/hld/host/arm32/lib -Wl,--no-as-needed -lalteracl -lalterammdpcie -lelf -lstdc++
```
是否添加的位置不对.我记得那个Makefile有很多ifdef. 这个貌似是x86的 link

tzechienchu, 2018-06-13:
Run on Windows  Quartus 17.1
C:\Users\>aocl link-config
/libpath:C:/intelFPGA/17.1/hld/host/windows64/lib OpenCL.lib

thinkoco, 2018-06-13:
@tzechienchu 
yes,  `aocl link-config` will get the x86 link. you can `set AOCL_BOARD_PACKAGE_ROOT ="${INTELFPGAOCLSDKROOT}/board/c5soc"` and then  run `aocl link-config --arm`.It will get arm link path

CrazyBingo, 2018-06-14:
@thinkoco @tzechienchu 
My point is, I have correctly set environment and verify in CMD just like blew:
[image]
[image]

CrazyBingo, 2018-06-14:
@thinkoco @tzechienchu 
[image]
It seems as that compile lib has not correctly linked in 17.1.. Still looking for the reason
...

CrazyBingo, 2018-06-14:
@thinkoco 
eventhough I mistaken set ‘INTELFPGAOCLSDKROOT’ as 'INTELFPGAAOCLSDKROOT' and modified it, aocl link-config is not correct. 
Emphasize that AOCL_BOARD_PACKAGE_ROOT variable is right, and I can correctly get 'aoc -list-boards'.
But the result is remain ... 
[image]
Otherwise， 16.1 & 17.1 board_env.xml is difference about lintel_soc32_mmd or altera_soc32_mmd

CrazyBingo, 2018-06-14:
@thinkoco @tzechienchu 
Finally, I done, along with many problems.
1) I have checked several time, that ensure my Environment Variable is correct！
[image]
2）Even if my environment is correct，I can not reach 'aocl link-config --arm'， it's still x86 lib. So Thks to @tzechienchu , I force set LINK like blow：
_LINK_CONFIG = $(wildcard $(INTELFPGAOCLSDKROOT)/host/arm32/lib/*.so) $(wildcard $(AOCL_BOARD_PACKAGE_ROOT)/arm32/lib/*.so)_
3）but after I make host，the compile tell me that can not found clGET** function, which is in host/include/CL, even Makefile have set CONFIG like blew:
_COMP_CONFIG = $(shell aocl compile-config --arm) -I$(HOST_INCS)  -DFPGA_DEVICE_
[image]
with no way out, I force COMP_CONFIG with *CL include, just like blow:
_COMP_CONFIG = $(shell aocl compile-config --arm) $(wildcard $(INTELFPGAOCLSDKROOT)/hld/host/include/CL/*.h) -I$(HOST_INCS)_ 
[image]
and then it passed.
[image]

4) Then I update the top.rbf/conv.aocx/run.exe to SDcard， source *17.1.sh， and .run.exe conv.aocx， and  the PipeCNN Passed。
[image]

Mark：
1）use 17.1 c5soc 'arm32/source/windows64/board_env.xml'， which is difference with 16.1, maybe not necessarily.
2）use scripts in c5soc/hardware/... that supprot 17.1(INTELFPGAOCLSDKROOT). user can use 16.1 script with 'ALTERAOCLSDKROOT'
3)）even though， should set variable 'ALTERAOCLSDKROOT'
..

Ths @thinkoco @tzechienchu , 
I think if we should use 17.1 Makefile format or not, because maybe solution is not the best...

thinkoco, 2018-06-14:
@CrazyBingo 
well done! I have no windows edition quartus for testing now. Also,you can try  running the `aocl link-config --arm` in in Soc_EDS_command_shell.bat  terminal  or running the init_opencl.bat before link-config. I'm not sure if it works. 
It's OK on Ubuntu PC. There is no error when compiling the run.exe for arm.
```
#export AOCL_ARCH_OVERRIDE=armv7l
export INTELFPGAOCLSDKROOT="/opt/intelFPGA/17.1/hld"
export QSYS_ROOTDIR="/opt/intelFPGA/17.1/quartus/sopc_builder/bin"
export QUARTUS_ROOTDIR_OVERRIDE="/opt/intelFPGA/17.1/quartus"
export PATH="$PATH:${QUARTUS_ROOTDIR_OVERRIDE}/bin:${QUARTUS_ROOTDIR_OVERRIDE}/linux64:${INTELFPGAOCLSDKROOT}/linux64/bin:${INTELFPGAOCLSDKROOT}/bin:${QSYS_ROOTDIR}"
export LD_LIBRARY_PATH="${INTELFPGAOCLSDKROOT}/host/linux64/lib:${AOCL_BOARD_PAKAGE_ROOT}/linux64/lib:${LD_LIBRARY_PATH}"
export AOCL_BOARD_PACKAGE_ROOT="${INTELFPGAOCLSDKROOT}/board/c5soc"
export QUARTUS_64BIT=1
export LM_LICENSE_FILE="/opt/intelFPGA/License.dat"
```

```shell
knat@knat-PC:~/terasic/PipeCNN/project$ aocl link-config  --arm
-L/opt/intelFPGA/17.1/hld/board/c5soc/arm32/lib -L/opt/intelFPGA/17.1/hld/host/arm32/lib -L/opt/intelFPGA/17.1/hld/host/arm32/lib -Wl,--no-as-needed -lalteracl -lintel_soc32_mmd -lstdc++ -lelf -lstdc++

knat@knat-PC:~/terasic/PipeCNN/project$ aocl link-config
-L/opt/intelFPGA/17.1/hld/board/c5soc/arm32/lib -L/opt/intelFPGA/17.1/hld/host/arm32/lib -L/opt/intelFPGA/17.1/hld/host/linux64/lib -Wl,--no-as-needed -lalteracl -lintel_soc32_mmd -lstdc++ -lelf
```

CrazyBingo, 2018-06-15:
Linux is an wonderful OS....TKS.  now I'am solve opencv problem now.
As you image support OpenCV-3.1.0, so I use opencv3.1.0 to compile Donny's PipeCNN, unfortunally，make host error with undefined reference to...
[image]

I do not know if I should give Windows...

thinkoco, 2018-06-19:
@CrazyBingo  您可以尝试直接在de1soc上编译host,这样应该比在windows上容易些.

thinkoco, 2018-06-26:
@CrazyBingo 
You can build the PipeCNN host with opencv libs on DE1SOC .
1. copy the PipeCNN source code to all_in_one_180317 sd card.
2. Power on DE1SOC with the all_in_one_180317 image sd card.
3. source init_opencl_17.1.sh
4. modify the Makefile like this:
[image]
5. make host

Also, you may change the `picture_file_path_head` path in main.cpp.

CrazyBingo, 2018-06-27:
@thinkoco 
Tks
First， I have solved it inwindwos，the reason why is that my arm-linux-gcc revision is altera iselsf, which is 4.8, is too low. It runs ok after I have install 4.9++.
Sensod, I will try to make host on DE10, TKS .

CrazyBingo, 2018-06-27:
@thinkoco 
Unfortunally:
[image]
while make host on de10-standard, it failed with no aocl command & CL/opencl.h file...

thinkoco, 2018-06-27:
@CrazyBingo yes, you may change the `char` to `signed char` at line 393 in layer_config.h

CrazyBingo, 2018-06-28:
@thinkoco 
TKS
[image]

closed 2018-07-13 by CrazyBingo

sergio14890, 2020-04-15:
HEY @CrazyBingo !
I have the same error you have.
What you do?