thinkoco/c5soc_opencl #10: Quartus about Build the Bsp, there is sth.

CrazyBingo, 2018-07-13:
Mr Thinkoco, some questions  about HowToBuildBSP.md，希望在有空的时候可以指导下，3KS
1）在添加8bit LED_PIO之前，其实你还添加了avalon-mm pipeline bridge, So how about the Pipeline's configuration(Width, Address and so on)
2）在第9点中，下面你还多了memory和periphral，这是什么时候添加的，以及如何互联总线

3）原始的iface也是个Qsys，是Terasic做的，你在他的基础上增加了VIP IP以支持VGA/HDMI功能，这一部分有没有相关的tourial
4）我竟然没有在Terasic的 CD中找到关于iface的制作的相关手册。
5）如果不用shareonly，而是ARM与FPGA的DDR/SDRAM分开，应该可以达到更高的带宽与效率，所以为什么都用shareonly
6）c5iface.ipx是个啥？

谢谢，我主要是弄裸奔FPGA硬件加速的，七八年前摸过SOPC后来在没弄过，这一块不太熟。

thinkoco, 2018-07-13:
@CrazyBingo 
1. 我使用的是terasic 16.0的BSP,在较新版本BSP的template中有这个bridge,用来挂载一些FPGA端的设备. 该bridge是挂载 Lightweight HPS-to-FPGA Bridge上的,所以外设地址=0xff200000 + 外设偏移地址.
2.memory是 HPS端DDR3的引线,peripheral 是HPS端外设引线,这个可在HPS配置,不用修改.
3.确实没有VIP添加的文档,二者类似,给出的文档包括了更新template所有需要的操作.其他修改也是一样的,无非是将新的逻辑添加到qsys或top.v. VIP的设置和连线可以参考DE1_SOC_FB.
4.OPENCL的bsp修改或定制,可能属于terasic技术支持外的范畴.也不排除后期terasic会添加这方面的文档.(一个庞大的工作)
5.shareonly 是对应单HPS DDR的板子的,对于HPS和FPGA端双 DDR的板子则可以用两种BSP
6.这个好像是用于指示acl_iface_system.qsys (opencl interface ) IP的路径.ip index file.

CrazyBingo, 2018-07-16:
@thinkoco TKS  我在看下  先把terasic的资料挖一挖 在弄这个。。
你弄VGA VIP是参考啥的。。  自己改的iface

thinkoco, 2018-09-21:
@CrazyBingo  参考的DE1_SOC_FB例子,在terasic的资料里面.

closed 2020-01-02 by thinkoco