thinkoco/c5soc_opencl #12: Error building SD Image

mdrolet01, 2019-08-11:
Hello,
Thank you for creating this nice repo.. it's been very helpful. I tried to build my sd image with the following command and it failed. I followed every step in your tutorial exactly, and I am building this with Ubuntu 18.04 for my DE1 soc. The script just hangs forever and never completes:

sudo ./make_sdimage.py -f -P preloader-mkpimage.bin,u-boot.img,num=3,format=raw,size=10M,type=A2  -P rootfs/*,num=2,format=ext4,size=1500M  -P zImage,u-boot.scr,opencl.rbf,socfpga.dtb,num=1,format=vfat,size=500M -s 2G -n sdcard_de1soc.img

Please let me know if you have any ideas on how to fix this.. thanks!

thinkoco, 2019-08-12:
@mdrolet01  yes, the python script works fine on ubuntu 16.04 , but get stuck on ubuntu 18.04.
This issue may be fixed by rsarwar87 [make_sdimage2.py](https://raw.githubusercontent.com/rsarwar87/altera-soc-rootfs/master/make_sdimage2.py)

mdrolet01, 2019-08-12:
That worked just fine, and my image was created. thank you!

closed 2019-08-12 by mdrolet01

jackfrye, 2019-12-21:
This page https://rocketboards.org/foswiki/Documentation/BuildingBootloader should be updated to reflect this. If anyone knows KeFu Fang, he should be informed.