thinkoco/c5soc_opencl #13: Possibility of extension to dual USB camera input

wei-lin-liao, 2019-09-03:
Hi thinkoco,
is it possible to modify driver or any way for dual USB camera application ?
Thanks a lot !!

thinkoco, 2019-09-10:
@wei-lin-liao  I don't know much about binocular cameras.Is it UVC  Standard Compatibility? If it is,there may be vedio0 and vedio1 in your /dev path. 
If the driver is special,you may need to port the driver first.

wei-lin-liao, 2019-09-13:
@thinkoco 
The dual USB camera means "2" USB camera, such as 2 set of Logitech webcam ( ex.C720 ). 
What I care about is "capability of usage of 2 set USB webcam at the same time".
After I trace the code about camera driver in the "video.cpp", it seems that video0 and video1 as what you say.
Thanks a lot !!