thinkoco/c5soc_opencl #15: @hammadj The BSP is OK , but there is something wrong with my sd card iamge. I have fixed this issue.U can run with the latest SD card image [here](https://drive.google.com/file/d/1uZkOzOZokQ7RGGJ8iyFA7wrLdqCmZTSu/view?usp=sharing) . Read the readme.txt in sd card first！

rahimijan, 2019-09-06:
@hammadj The BSP is OK , but there is something wrong with my sd card iamge. I have fixed this issue.U can run with the latest SD card image [here](https://drive.google.com/file/d/1uZkOzOZokQ7RGGJ8iyFA7wrLdqCmZTSu/view?usp=sharing) . Read the readme.txt in sd card first！
opencl runtime is 17.1
[image]

_Originally posted by @thinkoco in https://github.com/thinkoco/c5soc_opencl/issues/4#issuecomment-371899037_

rahimijan, 2019-09-06:
**@thinkoco** I downloaded ("#4 (comment) "),

but, its Linux doesn't boot up! In fact, I have just these pictures:

[image]
[image]

    Do you know what the problem is? and, can I ask you, if it is possible, test Linux of above link for me,

please, I need to work with it.

    Also, I want to install opencl on 'Linux LXDE', but I don't know How should I do it.

Please, if you know, help me to solve these issues.

Unfortunately, this Linux can't boot up from sd card (in DE10-Nano board), if you know the solution or another way to booting it up, Please let me know, thank you.

thinkoco, 2019-09-10:
@rahimijan The image has been tested on de10-nano version B2. Firstly, you can connect the uart to you PC, then power on the de10-nano. Also,you can try the  [latest iamge](https://drive.google.com/open?id=1r5qkYnTK9m7bwObakniFHgpD1JkJqDtD)
About building a sdcard image,here are some [guides](https://www.intel.com/content/www/us/en/programmable/documentation/ewa1403875738903.html#ewa1456771778778)

closed 2020-01-02 by thinkoco