thinkoco/c5soc_opencl #18: OpenCL SDK support for Quartus Prime Lite version

lutherstrike, 2019-12-10:


thinkoco, 2019-12-10:
Hi,@lutherstrike 
It 's OK on Ubuntu 18.04 with Quartus Prime Lite 18.1 and OpenCL SDK 18.1 (Standard).
[image]

Here is my environment-module files for environment variable,
```
  8 module-whatis "sets up access to  Quartus Prime 18.1 lite"
  9 #prereq 
 10 conflict altera
 11 global QSYS_ROOTDIR
 12 
 13 append-path PATH {/opt/intelFPGA_lite/18.1/modelsim_ase/linuxaloem};
 14 append-path PATH {/opt/intelFPGA_lite/18.1/quartus/bin};
 15 append-path PATH {/opt/intelFPGA_lite/18.1/quartus/linux64};
 16 append-path PATH {/opt/intelFPGA_lite/18.1/hld/linux64/bin};
 17 append-path PATH {/opt/intelFPGA_lite/18.1/hld/bin};
 18 append-path PATH {/opt/intelFPGA_lite/18.1/quartus/sopc_builder/bin};
 19 
 20 setenv QSYS_ROOTDIR {/opt/intelFPGA_lite/18.1/quartus/sopc_builder/bin};
 21 setenv QUARTUS_64BIT {1};

 23 setenv AOCL_BOARD_PACKAGE_ROOT {/opt/intelFPGA_lite/18.1/hld/board/c5soc};

 25 prepend-path LD_LIBRARY_PATH {/opt/intelFPGA_lite/18.1/hld/host/linux64/lib:/opt/intelFPGA_lite/18.1/hld/board/c5soc/linux64/li    b};
 26 setenv INTELFPGAOCLSDKROOT {/opt/intelFPGA_lite/18.1/hld};
 27 setenv QUARTUS_ROOTDIR_OVERRIDE {/opt/intelFPGA_lite/18.1/quartus};
 28 setenv TCL_LIBRARY {/opt/intelFPGA_lite/18.1/quartus/linux64/tcl8.6};

```

lutherstrike, 2019-12-10:
Thank you so much!
Intel makes it too hard to notice there is still a Stand-Alone SDK, they don't even mention that in any doc I can find from their official website...
I will try this.

closed 2019-12-10 by lutherstrike