thinkoco/c5soc_opencl #19: openCL FPGA SDK 17.1 De1SoC

Shahnewaz-rvis, 2019-12-30:
Hi, 
I am trying to use OpenCL SDK 17.1 (Quartus Standard ) with De1SoC
I am having a couple of problems 
OS Ubuntu 
Intel SDK installation 
Quartus Prime Lite 17.1  
Intel FPGA SDK for OpenCL 17.1  
SoC Embedded Design Suite (EDS)

1. Compiling host Application 
   I have installed all the necessary cross compiler using the command provided by your documents
sudo apt update
sudo apt install u-boot-tools gcc-arm-linux-gnueabihf g++-arm-linux-gnueabihf libncurses5-dev make lsb uml-utilities git

after issuing make host 

make host
arm-linux-gnueabihf-g++   host/src/color.o host/src/video.o host/src/AOCL_Utils.o host/src/screen.o host/src/opt.o host/src/main.o -o bin/camera_sobel.run -L/intelFPGA/17.1/hld/board/terasic/de1soc/arm32/lib -L/intelFPGA/17.1/hld/host/arm32/lib -L/intelFPGA/17.1/hld/host/arm32/lib -Wl,--no-as-needed -lalteracl -lalterahalmmd -lalterammdpcie -lelf -lstdc++  -lSDL2 
/usr/lib/gcc-cross/arm-linux-gnueabihf/5/../../../../arm-linux-gnueabihf/bin/ld: cannot find -lSDL2
collect2: error: ld returned 1 exit status
Makefile:115: recipe for target 'camera_sobel.run' failed
make: *** [camera_sobel.run] Error 1

**I already have installed sudo apt-get install libglib2 
the problem remains same** 

2.  Running c5soc_opencl_lxde_all_in_one_180317.img on De1Soc. 
After boot with de1soc (opecl rbf and dtd ) 
I executed sobel filter using RTE 17
Windows shows the picture 
pressing + / - / = does not have any effect. 

3.  if I run my opencl application .aocx it says 
libalterammd library missing 

Could please help me to sort out these problems, specially I am interested to 
run sobel using UVC camera using **intel fpga sdk 17.1  on DE1SoC**

Thanks a lot

thinkoco, 2020-01-02:
@Shahnewaz-rvis Hi, late reply

1. there is no sdl2, you can buid the opencl host on sdcard directly after boot the de1-soc. source the 17.1 runtime on terminal, install the libsdl2-dev libsdl2-2.0-0 and make host.
2.  it's "=" and "+" (shift =) nearby the backspace , not the numeric keypad
3. have more detail logs? have you source the run-time script firstly?

Shahnewaz-rvis, 2020-01-05:
Hi, 
Thanks 
I rebuild the Gray color application but has some problems. 
   1. UVC camera does not show the full image (camera provides 384x384 res.) 

 
However, I solved the cross compiler using the following  steps might be helpful for others,

**Solution regarding the compilation of HOST application for the board  DE1SoC**
**1. install Cross Compilation as follows** 
sudo apt install u-boot-tools gcc-arm-linux-gnueabihf g++-arm-linux-gnueabihf libncurses5-dev make lsb uml-utilities git

**2. Install SDL2 library on the cross compiler** 

hg clone https://hg.libsdl.org/SDL SDL
cd SDL
mkdir build
cd build
./configure --prefix=/home/ajay/intelFPGA/17.1/hld/host/arm32/  --disable-video-dummy --disable-video-qtopia --disable-cdrom  --disable-nasm --disable-alsa --disable-esd --disable-arts --disable-video-dga --disable-pulseaudio --disable-esd --disable-video-wayland --target=arm-linux --host=arm-linux
make
sudo make install
sudo ldconfig

**3. Then compile the host application**
go to the dir. 
cd /path/to/application/dir
make host

thinkoco, 2020-01-07:
@Shahnewaz-rvis Hi,
yes, cross compiling is a way to solve the issue. Sometimes, the version and the configuration  of library that you build may not be equal to those in the sd card. So, you can just build the host on de1soc. copy the source code to sd card, run de1soc, source the OpenCL runtime, cd to host dir and make host.(also, you can install  some libs by apt-get install on de1soc );
  About the resolution, you can test the camera on PC firstly by guvcview ,then you can get the supported resolution and format with you camera.
  In OpenCL host , the camera initialize by these code:
```
	video.format.type = stream_flag;
	video.format.fmt.pix.width = opt.width;
	video.format.fmt.pix.height = opt.height;
	video.format.fmt.pix.pixelformat = V4L2_PIX_FMT_YUYV;
```
  So, you can check the YUV 422 format with your camera.(V4L2_PIX_FMT_YUYV — Packed format with ½ horizontal chroma resolution, also known as YUV 4:2:2)

Shahnewaz-rvis, 2020-01-07:
HI,
Thanks a lot.  
I will have a look . Mean while I will be happy to get some example or directions
regarding 
1. How can I use De1SoC GPIO (head 0) from kernel code 
2. how I can use OpenCV acceleration mechanism in kernel code 

Thanks a lot

thinkoco, 2020-01-07:
Hi, @Shahnewaz-rvis 
1. [here](https://github.com/thinkoco/c5soc_opencl/blob/master/documents/HowToBuildBSP.md) is a way to add the PIO to the opencl BSP, and you can control it in opencl host.
2. you can install the opencv on sd card by apt-get ,but there is no mali gpu acceleration. Also, you may build the opencv with [neno acceleration](https://github.com/opencv/opencv/wiki/CPU-optimizations-build-options)

Shahnewaz-rvis, 2020-01-11:
Hi,
Thanks a lot. My intention is to run some specific algorithms of openCV on Cyclone V. Is there any resource exist regarding this?

thinkoco, 2020-01-19:
@Shahnewaz-rvis  sorry, I don't know the resources.