thinkoco/c5soc_opencl #20: De10-Nano doesn't boot with c5soc_opencl_lxde_all_in_one_180317.img

yurivict, 2020-01-19:
It doesn't connect to the network.

thinkoco, 2020-01-19:
@yurivict 
Maybe,  before  power on the de10-nano, the usb serial should been connected to PC usb port. you can do as [follows](https://github.com/thinkoco/c5soc_opencl#run-default-opencl-application) step by steps

kutenai, 2020-04-09:
@yurivict -- please check your subnets. My local network is on subnet 192.168.2.XX.
The arm linux only joins 192.168.1.xx or 192.168.0.xxx, so it was not connecting to the network.

`ifconfig` showed no eth0.
But, then I just configured it manually,

`ifconfig eth0 192.168.2.220`

and it worked.

Give that a try, see if it helps.

gflamis, 2022-03-20:
Hi @thinkoco,
I can configure dhcp or manually allocating address, this is fine. The problem I have is that I can transfer any files to DE10-nano via scp. Here is the command and the returned message...

_scp boardtest.aocx root@192.168.1.13:/home/root/working
root@192.168.1.13: Permission denied (publickey).
lost connection

I have verified firewall is disabled and I have also followed the procedure to generate and exchange ssh keys for the communication. 

Any help would be appreciated...
Giorgos