thinkoco/c5soc_opencl #21: Question about details of "Run default OpenCL Application"

wei-lin-liao, 2020-03-13:
Dear thinkoco,
In the step 
"**Copy the BOARD_NAME_APP.rbf and BOARD_NAME_socfpga.dtb in sdcard ( make sure the BOARD_NAME same as your target board) to Windows fat32 partition and rename them to opencl.rbf and socfpga.dtb**".
Q1. What is ther "Windows fat32 partition"? My personal desktop or the microSD card?
Q2. If the "Windows fat32 partition" were the SD card , I should replace the original "opencl.rbf" and "socfpga.dtb" with files which has the same name?

thinkoco, 2020-03-28:
@wei-lin-liao  sorry for late reply
1. on the microSD card
2.  for de10_nano target board
```
cp de10_nano/de10_nano_sobel.rbf  opencl.rbf
cp de10_nano/de10_nano_socfpga.dtb socfpga.dtb
```

wei-lin-liao, 2020-04-03:
@thinkoco 
Thanks a lot !
God bless Wuhan people !

closed 2020-04-03 by wei-lin-liao