thinkoco/c5soc_opencl #23: c5soc_opencl_lxde_fpga_reconfigurable.img -- partition size too small

kutenai, 2020-04-09:
I am using this image: c5soc_opencl_lxde_fpga_reconfigurable.img

But, the partition sizes are not very big, and the 2nd partition is the one with the Linux root FS on it, but there is a 3rd partition that has some required code on it (I can't remove it or the image won't boot at all), so I can't expand the root fs.

I'd like to re-build this image and re-order the partitions if possible so that the root fs is last.

Any thoughts, or information on who built this?

thinkoco, 2020-04-13:
Hi, @kutenai 

1. you can write the image to a micro sd card. delete the  sd card 3rd partition , then enlarge 2rd partition and make the 3rd partition (fat 32) on linux PC by  "gparted".
2. Or,  get the [files](https://drive.google.com/drive/folders/1B4fytaZv-4tmlPH5KmM2RPSX5v7qUhg4) , build the image and set the size like this  
```
sudo ./make_sdimage2.py -f -P preloader-mkpimage.bin,u-boot.img,num=3,format=raw,size=10M,type=A2  -P rootfs/*,num=2,format=ext4,size=4G  -P zImage,u-boot.scr,opencl.rbf,socfpga.dtb,num=1,format=vfat,size=500M -s 5G -n sdcard_de10_nano.img
```

kutenai, 2020-04-13:
The 3rd partition is partition type A2, which is uknown. I'm pretty sure this is the preloader, and the system won't boot without it. I'd have to have the preloader to re-build that partition.

What I'd really like to do is to re-build the rootFS and kernel from scratch, but no instructions for how that was built, or what versions of linux-sofcfpga were used.

It would be great if I know:
- what version of linux-socfpga was used to build kernel
- what this https://github.com/terasic/linux-socfpga, or the one in altera-opensource, or something else entirelly?
- What are the .config values used, i.e. the values used when doing `make menuconfig` for the kernel

How was the rootfs built? One thing I'd like to have are the build files used for making drivers, these are missing from the existing images, so no way to build a kernel installable driver.

thinkoco, 2020-04-13:
@kutenai
yes, you are right. the 3rd partition is for perloader and boot.
1.you can delete 1rd partition, expand the 2rd partition , then make 1rd partition again.
[image]

2 [here](https://github.com/thinkoco/c5soc_opencl/blob/master/README.md#how-to-do) are some guids for how to build  kernel , driver and sd card image

kutenai, 2020-04-13:
Yes, those are the instructions I need. I see that there is a special linux-sopcfpga fork under thinkoco

When I tried to `make zImage`, I get the following output
```bash
edhenderson@ubuntu:~/sdcard_build/thinkoco/linux-socfpga$ make zImage
scripts/kconfig/conf --silentoldconfig Kconfig
*
* Restart config...
*
*
* Support for small TFT LCD display modules
*
Support for small TFT LCD display modules (FB_TFT) [N/m/y] (NEW)

```

That is not expected since I copied over the .config as described here
`cp config_opencl_de10_nano .config`

I'm not sure why the .config is not accepted, but it seems to want to re-configure.
Any ideas?

kutenai, 2020-04-13:
I opened the .config file with `make menuconfig`, then just saved it. This got me past that initial prompt. It looks like the only difference is here:
```bash
edhenderson@ubuntu:~/sdcard_build/thinkoco/linux-socfpga$ diff config_opencl_de10_nano .config
1948a1949
> # CONFIG_FB_TFT is not set
```

thinkoco, 2020-04-14:
@kutenai 
have you install   gcc-arm-linux-gnueabihf  and set the ARCH  and CROSS_COMPLIE?

```
sudo apt install u-boot-tools gcc-arm-linux-gnueabihf g++-arm-linux-gnueabihf libncurses5-dev make lsb uml-utilities git

git clone https://github.com/thinkoco/linux-socfpga.git
cd linux-socfpga
git checkout -b socfpga-opencl_3.18 origin/socfpga-3.18
cp config_opencl_de10_nano .config

export ARCH=arm
export CROSS_COMPILE=arm-linux-gnueabihf-
export LOADADDR=0x8000

make zImage
```

kutenai, 2020-04-14:
Yes, i've done those things. I'm able to build now, so it's okay.
I'm making progress now. Thanks for your help.