thinkoco/c5soc_opencl #26: Open cl environment

18521613, 2021-03-17:
Hi all,
Could somebody tell me how can create De1-SoC linux image with init_opencl_17.1.sh file? I have spent a lot of time looking for any documents about it but i couldn't find anything!
Thanks all!

thinkoco, 2021-03-19:
> Hi @thinkoco
> Could you tell me how can create De1-SoC linux image with OpenCL driver and init_opencl_.sh file? I have spent a lot of time looking for any documents about it but i couldn't find anything!
> Thanks you!

Hi，

A:
 1.You can downlaod the c5soc_opencl_lxde_fpga_reconfigurable_20201027.img
2. do as follows，and choose de1-soc  in step 2
https://github.com/thinkoco/c5soc_opencl#run-default-opencl-application

Now, the sdcard is for de1-Soc with de1-Soc sharedonly-vga BSP

B:
For building your own sdcard,here are some guides.

https://github.com/thinkoco/c5soc_opencl/blob/master/documents/LinuxAndDrivers.md#how-to-update-opencl-driver

https://github.com/thinkoco/c5soc_opencl/blob/master/documents/HowToBuildSDImage.md

18521613, 2021-03-19:
hi @thinkoco,
I want to build my sdcard so i have tried to do your B option with socfpga-opencl_4.1.22 branch on your git. But when i use _cp config_opencl .config_ , the system informs that no file such a config_opencl or .config file. I have clicked on show hidden file but there is no file named config_opencl or .config

thinkoco, 2021-03-19:
@18521613
Hi, the socfpga-3.18  is a stable branch for opencl. 
Also, linux 4.9 and 4.19 has been tested with modified opencl driver.Here are some guides.
https://github.com/thinkoco/c5soc_opencl_rte/blob/master/documents/How_to_do.md

18521613, 2021-03-20:
@thinkoco I'm following your guides: https://github.com/thinkoco/c5soc_opencl_rte/blob/master/documents/How_to_do.md with linux 4.9. But when I'd used command _make zImage_, i got this error 
[image]

thinkoco, 2021-03-20:
hi @18521613 
have you check out the socfpga-4.9.78-aoc barnch ?
```
git clone --depth 3 --branch socfpga-4.9.78-aocl https://github.com/thinkoco/linux-socfpga.git
export ARCH=arm
export CROSS_COMPILE=arm-linux-gnueabihf-
export LOADADDR=0x8000
make c5socl_defconfig
make zImage -j4
make modules
make socfpga_cyclone5_de10_nano.dtb
```

18521613, 2021-03-21:
> hi @18521613
> have you check out the socfpga-4.9.78-aoc barnch ?
>
> ```
> git clone --depth 3 --branch socfpga-4.9.78-aocl https://github.com/thinkoco/linux-socfpga.git
> export ARCH=arm
> export [...]

Yes, I have do it but when use make _zImage -j4_ i got the same error

thinkoco, 2021-03-25:
@18521613 
for linux 4.9, no need to update files after checkout the ` socfpga-4.9.78-aocl` branch
```
git clone --depth 3 --branch socfpga-4.9.78-aocl https://github.com/thinkoco/linux-socfpga.git
export ARCH=arm
export CROSS_COMPILE=arm-linux-gnueabihf-
export LOADADDR=0x8000
make c5socl_defconfig
make zImage -j4
make modules
make socfpga_cyclone5_de10_nano.dtb
```
for linux 4.19, update the driver files  , makefile and kconfig

18521613, 2021-04-08:
Hi @thinkoco,
I have built my sd card image with opencl driver successfully. But when i try to execute hello_world example of BSP(Board Support Package) for Altera SDK OpenCL 18.1 from terasic (http://download.terasic.com/downloads/cd-rom/de1-soc/linux_BSP/OPENCL18.1/DE1-SoC_OpenCL_BSP.zip) i got this error:
[image]
Do you know the way to solve this error? Thanks you!

thinkoco, 2021-04-09:
@18521613 
maybe, the sdcard is unpluged  when data copying. just fix it by  `fsck`

tungdt113, 2021-04-15:
> @18521613
> maybe, the sdcard is unpluged when data copying. just fix it by `fsck`

Thanks for your help, I have solved that issue. But i got another issue :(. I have run `init_opencl_18.1_4.19.sh` but when i try to execute hello_world.aocx the system inform that _Couldn't open /dev/ac10 device! Did you load the driver?_
[image]

And when i run `ls /dev` there is no ac10 here! Do you know anything about this? 
[image]

Thank you!

thinkoco, 2021-04-15:
@tungdt113 
 Hi, the opencl driver should be rebuild when running with a new zImage. 
1. dowload the runtime enviroment files here. 
https://github.com/thinkoco/c5soc_opencl_rte/blob/master/linux-4.19.x/opencl_rte/aocl-rte-18.1.0-625.arm32.4.19.tar.xz
2. [here](https://github.com/thinkoco/c5soc_opencl/blob/master/documents/LinuxAndDrivers.md#how-to-update-opencl-driver) is the guide for buiding opencl driver
```
export ARCH=arm
export CROSS_COMPILE=arm-linux-gnueabihf-
cd  xxxxxx/aocl-rte-18.1.0-625.arm32.4.19/board/c5soc/arm32/driver
make KDIR=xxxxxx/linux-socfpga
```

18521613, 2021-04-19:
Hi @thinkoco,
I have built opencl driver and copy it to /home/root in partiotion 2 of sdcard, but it not working..

thinkoco, 2021-04-20:
Hi @18521613 
you may change the path in the `init_opencl_18.1_4.19 .sh` for your case , or copy it to `$AOCL_BOARD_PACKAGE_ROOT/arm32/driver/`
```
#!/bin/bash
export INTELFPGAOCLSDKROOT=/root/aocl-rte-18.1.0-625.arm32.4.19
export AOCL_BOARD_PACKAGE_ROOT=$INTELFPGAOCLSDKROOT/board/c5soc
export PATH=$INTELFPGAOCLSDKROOT/bin:$PATH
export LD_LIBRARY_PATH=$INTELFPGAOCLSDKROOT/host/arm32/lib:$AOCL_BOARD_PACKAGE_ROOT/arm32/lib:$LD_LIBRARY_PATH
insmod $AOCL_BOARD_PACKAGE_ROOT/arm32/driver/aclsoc_drv.ko
#export CL_CONTEXT_COMPILER_MODE_INTELFPGA=3
```

tungdt113, 2021-04-20:
@thinkoco,
I have load  aclsoc_drv.ko successfully but there is still no ac10 device here
[image]

tungdt113, 2021-05-31:
Hi @thinkoco,
i have tried to build OpenCL driver like this guide: https://github.com/thinkoco/c5soc_opencl/blob/master/documents/LinuxAndDrivers.md. I have downloaded Intel FPGA Runtime Environment for OpenCL Linux Cyclone V SoC TGZ 17.1 on Altera Download Website. And I build it for kernel 3.18 like your image: _c5soc_opencl_lxde_fpga_reconfigurable_20201027.img_. But when I try to test my OpenCL driver on your image. I got this error like this:
[image]

tungdt113, 2021-06-17:
Hi @thinkoco 
I've successfully generate "/dev/acl0" when source init_opencl.sh file. But when i try to execute the hello example from terasic. I got this error.
[image]
i've checked /sys/class/fpga_bridge folder there is no folder have name like them
[image]
But when i've checked /sys/class/fpga_bridge in your linux image, it has the above folders. So could you tell me how to generate them? 
Thank you!

18521613, 2021-06-18:
> @thinkoco I'm following your guides: https://github.com/thinkoco/c5soc_opencl_rte/blob/master/documents/How_to_do.md with linux 4.9. But when I'd used command _make zImage_, i got this error
> [image]

@thinkoco have you solved this error, yet?

thinkoco, 2021-06-18:
@18521613  
the error occurs when using gcc 5.0 and above, it has been fixed. you can checkout the lastest code on branch `socfpga-4.9.78-aocl`

tungdt113, 2021-06-19:
@thinkoco,
I've rebuild zimage on `socfpga-4.9.78-aocl` branch, it builds successfully. But when boot this zImage and socfpga.dtb for de1_soc that i take from in your image : `c5soc_opencl_lxde_fpga_reconfigurable_20201027.img` , it seemly don't add /dev/fpga0 char driver and file operations.
[image]
And i have a question that if i can use `socfpga_cyclone5_de1soc.dts` from `socfpga-3.18` branch to build `socfpga_cyclone5_de1soc.dtb` for `socfpga-4.9.78-aocl`?? When i use this dtb to boot, it stop booting kernel like this:
`socfpga-4.9.78-aocl`
[image]

tungdt113, 2021-06-19:
@thinkoco
i have a question that if i can use `socfpga_cyclone5_de1soc.dts` from `socfpga-3.18` branch to build 'socfpga_cyclone5_de1soc.dtb` for `socfpga-4.9.78-aocl` ??

tungdt113, 2021-06-19:
@thinkoco ,
I've rebuild zimage wit

> @18521613
> the error occurs when using gcc 5.0 and above, it has been fixed. you can checkout the lastest code on branch `socfpga-4.9.78-aocl`

@thinkoco,
I've rebuild with your update, it builds successfull. But when i boot this `zImage` with `socfpga.dtb` for de1soc that i take from your image: `c5soc_opencl_lxde_fpga_reconfigurable_20201027.img`. It seemly don't add /dev/fpga0 char driver and file operations. And i saw you change something in `c5soc_opencl_rte/blob/master/socfpga-4.9.78-ltsi/include/linux/fpga/fpga-mgr.h`, but i did'n see any change in` fpga-mgr.h` of `socfpga-4.9.78-aocl` branch

thinkoco, 2022-05-05:
@tungdt113 
yes, it’s [here](https://github.com/thinkoco/linux-socfpga/commit/cb24d83e6ef8d742d8a396248c83ececff5fff89)