thinkoco/c5soc_opencl #30: Could add open driver in img

tungdt113, 2022-05-05:
Hi @thinkoco,
I want to open some driver in your image `c5soc_opencl_lxde_fpga_reconfigurable_20201027.img`. There is any way to do that without remaking the Image?
Thank you!

tungdt113, 2022-05-05:
And I also got a problem when building an image for arria 10 with `socfpga-opencl_4.1.22` repo ( I know your repo mainly supports for cyclone 5 but i think maybe you can know about this). When I run aocl I got the below error.
[image]
The image seemly don't add /dev/fpga0 char driver. It is the same as the last issue I've asked you https://github.com/thinkoco/c5soc_opencl/issues/26. Do you know about this issue? 
Thank you!

thinkoco, 2022-05-05:
@tungdt113 
the default kernel is 3.18 in c5soc_opencl image. 

1. you can build your driver base on the 3.18  kernel. and force insmod  your driver. xxx.ko. 
2. Or, build an new zImage that include your driver , and rebuild the opencl driver. then update the zImage and driver.  [Here] (https://github.com/thinkoco/c5soc_opencl/blob/master/documents/LinuxAndDrivers.md#how-to-update-opencl-driver) is the guide 

About arria10 soc，not tested. It seems that the opencl driver is different form cyclone V soc. 
[Here](https://www.intel.com/content/www/us/en/docs/programmable/683788/19-1/building-the-software-and-sd-card-image.html#dropdown-1-2) is the guide for arria 10 soc

tungdt113, 2022-05-10:
Hi @thinkoco,
I am trying to build driver for UVC base on the 4.1.22 kernel and force insmod for my uvcvideo.ko but I got this error below `insmod: ERROR: could not insert module uvcvideo.ko: Unknown symbol in module`.
Do you know how to solve this problem ?
Thank you.

thinkoco, 2022-05-10:
Hi @tungdt113 
So，you have a golden altera soc opencl sdcard image base on Linux 4.1.22 and want to install UVC driver ?  
run demsg to find the unknown symbol name and find which driver contains the symbol . install it before uvcvedio.ko. Or find the dependence in the  module.dep file that generated when  you building uvc driver.

tungdt113, 2022-05-11:
Hi @thinkoco,
1. The Alter SoC OpenCL sdcard image base on Linux 4.1.22 i got from  `/home/thanhtung/intelFPGA_pro/18.1/hld/board/a10soc/a10soc_linux_sd_card_image.tgz`
2. I reused the zImage, the rootfs is built base on your guide with ubuntu 18.04.
3. And I succeed to build a new SoC OpenCL sdcard image but the zImage that I reuse has no UVC driver.
4. So I built UVC driver and force insmod for my uvcvideo.ko that I have discussed with you. Although I find the dependence in the `module.dep`, I still got this bellow error.
[image]
5. This is the content of `module.dep` file
[image]
6. Command that I force insmod
[image]

Thank you

thinkoco, 2022-05-11:
hi @tungdt113 
It seems that these driver need some unknown symbols that are not included in this zImage.  you can use ‘dmesg’ to find the symbol，and  check if enable some config define that includes these symbols or not.  If a define include the unknown symbol , then remove the config define when building the uvc driver.  This solution may be useless sometimes. 
 
Another way is the official guide.
https://www.intel.com/content/www/us/en/docs/programmable/683788/19-1/building-the-software-and-sd-card-image.html#dropdown-1-2
you can build zImage，dtb and opencl driver yourself.

tungdt113, 2022-05-17:
Thanks for your help. I will close this issue.